GrapheneOS/Camera #14: add support for RAW once CameraX supports it

thestinger, 2021-10-05:


thestinger, 2022-02-28:
Once this is implemented, we can consider implementing saving in other formats like AVIF when RAW support is available by converting from the bitmap provided by the RAW format.

Zanthed, 2022-05-17:
We're not a Camera2 app. We're a CameraX app.

Silent-Hunter, 2022-05-17:
My mistake, I thought X was like a placeholder for a number.

Zanthed, 2022-05-17:
CameraX is the abstraction of Camera2 and even Camera1 APIs made with far better integration, ease of use, portability, and even compatibility with legacy / broken devices instead of having the developers themselves hardwire quirks and workarounds and use very complex APIs that come with Camera2/1.

https://developer.android.com/training/camerax

Silent-Hunter, 2022-05-17:
I see! I noticed OpenCamera has both legacy camera and Camera2 options. But I understand not wanting to have both CameraX _and_ Camera2 APIs in the same app.

thestinger, 2022-05-17:
@Silent-Hunter You're misunderstanding something.

MHShetty, 2024-06-12:
Here's a feature request filed on CameraX's issue tracker related to this issue - https://issuetracker.google.com/issues/177918194

MHShetty, 2024-09-06:
The CameraX team has accepted this issue and we should have something that can be used to get access to RAW/DNG to support custom formats and requirements in their upcoming releases

FLchs, 2024-11-04:
🎉
> RAW/RAW + JPEG implementation is already done in CameraX. It will be released in the next version (1.5.0 alpha03).

https://issuetracker.google.com/issues/177918194#comment12

FLchs, 2025-03-05:
Hello, any update on this issue ?