GrapheneOS/Camera #47: Expand settings button

cloudcoroner, 2021-11-01:
While testing the setting last night, I noticed the expand button that is visible when you first open the app disappears while in the settings dialogue/menu. It appears when you change from 4:3 to 16:9 due to a display refresh. In keeping with the android expand look, should the button show ^ while the user is in settings?

This did not impact the functionality of the settings menu in my testing.

MHShetty, 2021-11-01:
Hi @cloudcoroner,

From what I am able to understand, the issue here is that the settings dropdown icon reappears when we toggle an option that causes the camera to restart (for e.g. the aspect ratio in your case). After some recent changes that were previously made, the settings icon no remains visible when the settings dialog is open (even if the camera restarts), so having something like ^ wouldn't really make any difference now.

Hope this resolves your issue.

Please do check the latest if it does.

Thanks for reporting this issue!

cloudcoroner, 2021-11-01:
no, its not really a problem with changing the aspect ratio. I was just reporting that the options/setting button disappears when the options window is open or expanded. I was thinking it should remain visible but change to a ^ button so the user can collapse the window
[image]
[image]
.

MHShetty, 2021-11-01:
The settings icon disappearing when the settings dialog is visible is the default/expected behavior.

cloudcoroner, 2021-11-01:
OK, just noticed it and wanted to make sure!

closed 2021-11-01 by cloudcoroner