GrapheneOS/Camera #51: Add setting option for video rate

emersion, 2021-11-05:
Some cameras (e.g. Pixel 3a) support video capture up to 240Hz (with a lower resolution). This is useful in some niche use-cases, e.g. timing events (physics, electronics, etc).

thestinger, 2021-11-06:
I think this depends on CameraX adding support for it. Ultra high frame rate modes may require a CameraX extension instead of simply changing the rate. Not sure about that.

MHShetty, 2021-11-06:
I have made a feature request for this, it's status can be tracked from here: 
https://issuetracker.google.com/issues/205334816

Donkey-Doug, 2022-04-18:
What about resolution? My phone (Xiaomi Mi 9T Pro) only films in 1080p @ 30fps.

MHShetty, 2022-05-24:
Hi @Donkey-Doug,

> What about resolution? My phone (Xiaomi Mi 9T Pro) only films in 1080p @ 30fps.

We might need to wait until the CameraX team includes this in their public schedule or any of their future release to be able to actually alter these configurations from our end. However, we must bare in mind that CameraX is library that's an abstraction to Camera2 and other related APIs that prima, so it can't really alter the hardware limitations of any device as such (or at least has it's own limits of doing so). 

Specifically speaking of Xiaomi Mi 9T Pro, it seems to only support 1080p (Full HD) at 30 FPS at best, so I don't really think our camera app would be able to reliably capture videos at a higher FPS than that (for example, at 60 FPS) unless the CameraX team somehow comes up with a workaround (which is however highly unlikely considering the ton of workload and feature requests they already have).

We'll add an update to this thread as soon as CameraX includes an option to alter the frame rate/resolution and we include it in our camera app.

Thanks a lot for trying out our camera app @Donkey-Doug!

MHShetty, 2022-05-24:
> What about resolution? My phone (Xiaomi Mi 9T Pro) only films in 1080p @ 30fps.

We might need to wait until the CameraX team includes this in their public schedule or any of their future release to be able to actually alter these configurations from our end. However, we must bare in mind that CameraX is library that's an abstraction to Camera2 and other related APIs that prima, so it can't really alter the hardware limitations of any device as such (or at least has it's own limits of doing so). 

Specifically speaking of Xiaomi Mi 9T Pro, it seems to only support 1080p (Full HD) at 30 FPS at best, so I don't really think our camera app would be able to reliably capture videos at a higher FPS than that (for example, at 60 FPS) unless the CameraX team somehow comes up with a workaround (which is however highly unlikely considering the ton of workload and feature requests they already have).

We'll add an update to this thread as soon as CameraX includes an option to alter the frame rate/resolution and we include it in our camera app.

Thanks a lot for trying out our camera app @Donkey-Doug!

r3g-5z, 2023-04-19:
CameraX now supports this as of [1.3.0-alpha06](https://developer.android.com/jetpack/androidx/releases/camera#1.3.0-alpha06)

> Add setTargetFrameRate() API in the VideoCapture.Builder and getTargetFramerate() API in VideoCapture

octocorvus, 2023-04-20:
A public API to query supported frame rate ranges for device was also added in this update. From https://developer.android.com/reference/androidx/camera/core/CameraInfo#getSupportedFrameRateRanges()
> There is no guarantee that these ranges can be used for every size surface or combination of use cases. If attempting to run the device using an unsupported range, there may be stability issues or the device may quietly choose another frame rate operating range.

For now, maybe we can add an option to prefer either 60fps or 30fps, and make it clear that this setting might not work for all resolution settings.

Thovthe, 2025-04-14:
Hey what's the status of this?  Looks like there may be a path to getting this working with the information here and in the PR.

MHShetty, 2025-04-14:
Hi @Thovthe,

We have the code ready and tested for setting the video frame rate at #436. We're mainly just waiting for the CameraX team to fix a certain [issue](https://issuetracker.google.com/issues/384146331)that's present with the API before we're able to merge it with our camera app

Thovthe, 2025-04-14:
Good to see.  Hopefully the upstream Android development changes don't interfere with a speedy fix.  If it looks like this issue has been deprioritised on their end would you consider including this behind an experimental features checkbox or something like that?  Thanks for the response.