GrapheneOS/Camera #52: 16:9 Display Can't Adjust In Camera And Video Mde

IzumiSena, 2021-11-05:
[image]
[image]
[image]

Its Happened Only if i use 16:9!
Works Fine in 4:3!
4K video recording add so much glitch! ... And if o try to see 4k recorded video it's laggy!.
(Sorry for my bad English).

MHShetty, 2021-11-05:
We only support devices that are officially supported by GrapheneOS.
https://grapheneos.org/faq#device-support

This issue seems to be on the end of the CameraX library as we aren't really controlling the low level stuff while recording a video. We have reported this issue to them and this should automatically get resolved once they fix it from their end and make a release that includes those changes and we upgrade to the latest/that version of the library.

The status of the issue can be tracked from here:  https://groups.google.com/a/android.com/g/camerax-developers/c/P6DORRXqspk

Thanks for reporting this issue!

closed 2021-11-05 by MHShetty

MHShetty, 2021-11-06:
You can track the status of this issue from here: https://issuetracker.google.com/issues/205340278

Once fixed, the issue should get resolved in the next release of the library. 

This might take a while as there hasn't been a release of the main library since a really long time.

MHShetty, 2021-11-17:
@IzumiSena The issue with those colored lines have now been fixed by the CameraX team: https://issuetracker.google.com/issues/205340278

Hope they include the changes in the next release soon!

Thanks for filing this issue @IzumiSena!

MHShetty, 2021-11-19:
The colored lines issue has now been fixed by the CameraX team and the width should fit better in the next release.

Thanks a lot for reporting these issues!