GrapheneOS/Camera #53: add zoom to in-app gallery

thestinger, 2021-11-07:


reinold-p, 2021-12-08:
+1: would be very helpfull to control details after taking a foto ...and to be able to delete it immediately

MHShetty, 2021-12-09:
Hi @reinold-p,

Controlling details of an image captured by the camera app and being able to delete it immediately (unsure what this means), seem to be separate issues/feature requests to me. Please file an issue/FR for each of those recommendation/suggestion(s) describing them in a bit more detail (as to what is exactly required/expected).

Thanks a lot for testing our app!

MHShetty, 2021-12-09:
This enhancement has now been added to our app in the latest changes pushed to the repository (Commit ID: 4ae886162b7f6271b5fda4ca7e2cdb86ee4d9335).

closed 2021-12-09 by MHShetty