GrapheneOS/Camera #56: Save selfie as previewed (Add)

bugycugy, 2021-11-07:
**Would you consider adding this feature?**

"Save pictures with the same orientation seen in the viewer."

thestinger, 2021-11-07:
Flipping the image via EXIF is entirely possible just like how the orientation is set based on the device's rotation. It's entirely possible to do that on your own for the time being. The captured image is always in the same actual orientation with the metadata determining how it's shown.

bugycugy, 2021-11-07:
> Flipping the image via EXIF is entirely possible just like how the orientation is set based on the device's rotation. It's entirely possible to do that on your own for the time being. The captured image is always in the same actual orientation with the metadata determining how it's shown.

I know the details but this feature comes in handy, I like your camera app very much. I am using it on my Mi Note 10 Pro (OctaviOS - A12). In my opinion, this is the only missing feature, so I wanted to convey it to you.

closed 2021-11-10 by MHShetty

MHShetty, 2021-11-10:
A persistent option has been added to prevent the flipping of image.  The changes have been pushed to the repository and the option should be visible for the front-camera in the next release.

Thanks a lot for requesting this enhancement @bugycugy!

bugycugy, 2021-11-25:
> A persistent option has been added to prevent the flipping of image. The changes have been pushed to the repository and the option should be visible for the front-camera in the next release.
> 
> Thanks a lot for requesting this enhancement @bugycugy!

Thank you very much for adding this feature...

commit dbd7edf1 (2021-11-10): Add persistent option to save image as previewed
(Fixes #56)