GrapheneOS/Camera #57: Camere's navigation is off center

JanOstrowka, 2021-11-07:
[image]

When using the newly released camera app, I've noticed that navigation tabs ("QR SCAN", "CAMERA", "VIDEO") never seem to align properly and tend to float off center. Most of the time, when switching mode, tabs align in the center and then move abruptly, to the wrong place.

It is a very minor bug. I also would like to say, that this is a really welcome update, that has finally made the native Graphene OS camera usable. Thank you!

---

**Hardware**:
os: 2021110617
phone: Pixel 5

MHShetty, 2021-11-08:
Hey @JanOstrowka,

The bottom tab layout needs to be worked upon/full re-designed in order to resolve this and some other issues related to it. We are working on this and will revert as soon as this gets fixed.

Thanks a lot for reporting this issue!

bd0n4lds, 2021-11-09:
I see the same on different hardware.
Hardware:
os: 2021110617
phone: Pixel 4a

MHShetty, 2021-11-09:
> I see the same on different hardware. Hardware: os: 2021110617 phone: Pixel 4a

This isn't a device/hardware specific issue. Any updates regarding this issue shall be posted here.

MHShetty, 2021-11-14:
Hi @JanOstrowka,

This issue has been resolved in the latest changes pushed to the repository. Changes should be visible in the next release. 

Thanks a lot for reporting this issue!

commit 233e1484 (2021-11-14): Fix issues with centering and snapping of bottom tab layout
(Fixes #57)
closed 2021-11-14 by MHShetty