GrapheneOS/Camera #61: Record video  as previewed

lmJojo, 2021-11-12:
The video I recorded with my front-facing camera are always flipped. How can save it as previewed?

thestinger, 2021-11-12:
You can do this by altering the metadata. Our Camera app doesn't support flipping it in the metadata yet.

MHShetty, 2021-11-12:
Currently CameraX only supports altering metadata for the images it takes and unfortunately not video (the same reason why geo-tagging isn't supported for video yet). We might need to wait until they make the required changes and include it in one of their releases.

joklmT, 2021-11-13:
Thank you for your answers. I get it.

MHShetty, 2022-05-24:
Hi @lmJojo,

A small update related to this feature request -

The CameraX team seems to have considered this feature request (that was earlier made by another user) -
https://issuetracker.google.com/issues/194634656#comment9

We'll add this feature to our camera app as soon as CameraX supports it in one of their future releases.

Thanks a lot for making this feature request @ImJojo!

GurucharanSavanth, 2022-05-24:
Thanks

-- Sent from /e/ Mail.

r3g-5z, 2023-04-19:
CameraX now supports this as of [1.3.0-alpha06](https://developer.android.com/jetpack/androidx/releases/camera#1.3.0-alpha06)

> Added VideoCapture mirroring APIs, including VideoCapture.Builder.setMirrorMode(int) and VideoCapture.getMirrorMode(). The APIs are useful for applications require the video recording to be consistent with common camera preview behavior, i.e. the rear camera preview is not mirrored but the front camera preview is mirrored.

lmJojo, 2023-04-20:
Thanks you.I will try it.

thestinger, 2023-04-20:
We haven't added the feature yet.

lmJojo, 2023-04-20:
Ok.I'm going to try it in my demo

MHShetty, 2024-09-06:
The issue has been fixed in the PR mentioned above (#435)

closed 2024-09-06 by MHShetty