GrapheneOS/Camera #62: Camera: "Unlocked camera" settings are different from "locked camera" settings

camoz, 2021-11-14:
Hi,

I noticed that the camera settings when used in normal "unlocked" mode are not propagated to the camera when accessed via the power-button-double-tap gesture while the phone is locked locked, and vice-versa. They seem to maintain their own set of settings. Tested with the settings "Camera sounds" and "4:3 vs 16:9".

Not sure if this is intended behavior. But it seems to be suboptimal UX.

I have GrapheneOS SP1A.211105.004.2021110617 installed on a Pixel 5.

thestinger, 2021-11-14:
It's intentional that changes made in the special activity while locked aren't done for the full unlocked activity.

thestinger, 2021-11-14:
There's some room for making different decisions about the details beyond that, but that part is important.

MHShetty, 2021-11-17:
This is the intended behavior and is mainly done for user privacy. We might add a way to toggle this behavior in the future if we add an extended settings activity as suggestions/requests for having more options come in (to avoid cluttering the settings dialog with too many options). Please do feel free to share what you think about this.

Thanks for filing this issue @camoz!

closed 2021-11-17 by MHShetty

camoz, 2021-11-17:
Ok, yes I already thought this is intended because of the security implications.

Personally I don't find this annoying. It was just confusing learning about it without prior knowledge: I disabled the camera shutter sound in the app, then later that day took a photo in locked mode, where it of course *did* make a sound. My immediate reaction was "oh no, probably a bug, damn, I hate this sound". Then by experimentation I found out that both modes retain their settings, but in different states.

Maybe a little note about that intended behavior when opening the camera the first time would help, so that it does not look like a bug at first. An alternative would be explaining it on the usage webpage.

Not sure if it is worth spending much effort to implement a way to toggle this behavior. Configuring one's preferred camera settings twice does not seem like a big issue, since there are not many settings. At least, I think for now GOS has more important things to work on, than optimizing that bit of UX. But documenting the behavior somewhere on the website (or even on the phone) as mentioned above would be helpful I guess.