GrapheneOS/Camera #65: Save on mirco sd

qw123wh, 2021-11-20:
Can you add the ability to save photos / videos on micro sd?

lazytownfan, 2021-11-27:
This would be nice, if testing this app on non-GrapheneOS devices that still have micro SD card slots (or any alternative storage location besides directly on the device's built-in storage).

I'm aware that this Camera app is officially only for Pixel devices currently supported by GrapheneOS, though sometimes I sideload this app and test it on the Samsung Galaxy Tab S6 Lite (`gta4xlwifi`) as an example.

thestinger, 2021-11-27:
Replacing with https://github.com/GrapheneOS/Camera/issues/72 since this won't be implemented with any particular hard-wired locations. That's not actually possible with how modern shared storage support works.

closed 2021-11-27 by thestinger