GrapheneOS/Camera #66: FR: Add configuration option to set custom filename format

Starkstromkonsument, 2021-11-20:
Pictures are currently saved with filenames of the format  `IMG_%Y%m%d_%H%i%s.jpg` (e.g. `IMG_20211114_191156.jpg`). Please add an configuration option to allow specifying custom filename formats. I personally prefer the format `%Y%m%d_%H%i%s.jpg` (e.g. `20211114_191156.jpg`). Thank you.

MHShetty, 2021-11-24:
Hi @Starkstromkonsument,

Could you please share all the file name formats you'd like the app to support or will having an textfield for the same be better?

Starkstromkonsument, 2021-11-24:
I'm fine with `%Y%m%d_%H%i%s.jpg` but a textfield would be a much more flexible and sustainable approach. Perhaps providing a drop down menu with some default formats and the abbility to add custom formats would be great.

barthermans, 2024-04-26:
I would like this feature as well. The IMG and VID prefixes don't really make sense (to me) and it messes up the sorting order of the files.
Maybe an option can be added to enable/disable the prefix?

mumbo2030, 2026-09-08:
I also find the default naming prefixes a bit annoying when trying to organize my photos once they are off the phone. I currently have to undergo a manual process of renaming so being able to change this would be a nice QoL bump.