GrapheneOS/Camera #68: The UI is off-center after v4

flexagoon, 2021-11-22:
Here's how the UI looked before v4:
[image]
And here's what it looks like now:
[image]
It's off center and there's a lot of unused space in the bottom

thestinger, 2021-11-22:
On which device?

flexagoon, 2021-11-22:
> On which device?

I have OnePlus 7T Pro.

closed 2021-11-23 by MHShetty

flexagoon, 2021-12-01:
@MHShetty unfortunately the issue still persists in v7

thestinger, 2021-12-01:
Show a screenshot.

flexagoon, 2021-12-02:
[image]
Its pretty much the same as it was

flexagoon, 2022-01-06:
@thestinger @MHShetty could you please re-open this issue? I still have it.