GrapheneOS/Camera #71: Enhancement - Add exposure and focus lock functionality

snrkl, 2021-11-25:
Many camera applications support the ability to press a point in the compose photo frame to spot focus and meter.

If the user presses and holds however, the exposure and focus can be locked, and prevent any automatic focus or exposure controls until the user taps to focus, or takes a picture. 

It would be useful for the Camera App to provide mechanism to nominate a spot focus and exposure point, and locking it.

MHShetty, 2021-11-25:
Hey @snrkl,

I'll need to check if CameraX supports this and just in case if it does then I'll soon implement it 👍

Thanks a lot for making this feature request!

MHShetty, 2021-12-01:
Hey @snrkl,

Focus lock can be achieved by setting the focus timeout to Off/0... For exposure lock, the UI is ready, just need to figure out a way to actually implement it via CameraX. If it's possible to implement this, this option should be visible in v8.

Thanks a lot for making this feature request!

thestinger, 2021-12-01:
That will also lock exposure and white balance since tap to focus does auto focus + auto exposure + auto white balance, not just focus.

MHShetty, 2021-12-18:
Hey @snrkl,

CameraX currently doesn't directly offer support for exposure/white-balance lock, so we might need to wait for a while before they actually support.

MHShetty, 2025-07-12:
Reference of CameraX issue: https://issuetracker.google.com/issues/208583141 (to know status of the feature request)