GrapheneOS/Camera #75: Add support for "Motion Photo" jpeg.

lbdroid, 2021-11-30:
GCam creates a small video around the shutter moment in mp4 format and appends it to the end of the jpeg file. They also add a few EXIF tags describing the video, example as follows;

```
Motion Photo                    : 1
Motion Photo Version            : 1
Motion Photo Presentation Timestamp Us: 1166199
Directory Item Mime             : image/jpeg
Directory Item Semantic         : Primary
Directory Item Length           : 0
Directory Item Padding          : 0
Directory Item Mime             : video/mp4
Directory Item Semantic         : MotionPhoto
Directory Item Length           : 3774306
Directory Item Padding          : 0
```

The video, in this case, is 3,774,306 bytes at the end of the file.
The format of the video is an *actual* mp4 video. It can be copied into an mp4 file directly using "dd" as;
```dd bs={filelen-3774306} skip=1 if=file.MP.jpg of=file.MP.mp4```

Zanthed, 2021-11-30:
Just to add a bit of information in case anyone's confused:

This is the equivalent to Apple iOS's "Live Photo" pictures.

MHShetty, 2021-12-01:
We could add this as a separate mode for having motion photos in our camera app. I'll try to add this to our app maybe after the upcoming release (if it is possible).

realpixelcode, 2022-03-24:
How could one view such motion photos on ungoogled devices?

realpixelcode, 2022-03-24:
@lbdroid Do you know of any OSS apps that have such a viewer included?

lbdroid, 2022-03-25:
@realpixelcode : That's out of scope. This issue is about CREATING the files.

realpixelcode, 2022-03-25:
Sure, but why bother creating fancy motion photos if the user isn't even able to view them in the first place (or has to rely on proprietary Google apps)?

lbdroid, 2022-03-25:
1 step at a time. Feel free to create another issue for that. Please stop polluting this issue.

thestinger, 2022-03-25:
The camera's gallery would need to be able to display them.

realpixelcode, 2022-03-25:
> The camera's gallery would need to be able to display them.

Yeah, exactly. Otherwise, it wouldn't make sense to implement motion photos.

lbdroid, 2022-03-25:
@realpixelcode : No, you completely missed the point. The camera's gallery is IN THE CAMERA APPLICATION.

realpixelcode, 2022-03-25:
Yes. I know that. Graphene camera's internal gallery must be able to view motion photos. Otherwise it does not make sense to implement the option to take motion photos.

lbdroid, 2022-03-25:
Are you quite finished now that you've contributed absolutely nothing?

thestinger, 2022-03-25:
@lbdroid Why are you doing this across a bunch of issues?

lbdroid, 2022-03-25:
This is an issue that I brought forward, you think its inappropriate to be slightly defensive when someone tries to side track it?

sethgillett, 2022-04-29:
OP's apparent toxicity aside, it may be simpler and more practical to simply capture an mp4 file alongside the jpeg file (the two files would have to have the same name and exif data). Many photo galleries (apple's photos app, photoprism, and likely google photos) have learned to recognize this format as it is what apple uses for its "live photos" feature. The internal gallery would simply need to load and play the video whenever a user long presses the image, and not load them as two separate files side by side.