GrapheneOS/Camera #78: Add support for portrait mode

flawedworld, 2021-12-11:


MHShetty, 2021-12-11:
Hey @flawedworld,

The app already indirectly supports portrait mode in in-app gallery and more settings activity. For the main camera activity, we are just rotating the icons based on the orientation of the device (since natively supporting this could cause the camera to restart [requires a short delay] which could be bad while recording too [would cancel it]). The output image/video(s) respect the device's orientation so that wouldn't be a problem too.

If you're referring to the **Portrait** mode found in GCam, then this should get implemented as soon as extension modes support come out for Pixel devices. (I'll just keep the last paragraph for future issue filers for reference if that's the case)

Thanks for filing this issue @flawedworld!

Zanthed, 2021-12-11:
Likely referring to the Portrait mode that uses special image effects make a self-portrait of yourself or something stand out more, e.g. depth of field.

MHShetty, 2021-12-11:
Actually the app already does support vendor specific extensions, so any device that has exposed their portrait vendor specific extension will automatically have that mode. Since we're mainly targeting Pixel devices, we might need to until it gets exposed for Pixel devices.

MHShetty, 2021-12-18:
Hey @flawedworld, since the app already supports all the CameraX vendor specific extensions (including portrait mode), this issue might have already been resolved. We might need to wait for Pixel devices to officially expose those extensions to be able to view them on the devices we support.

I'll be closing this issue for now.. Please feel free to re-open this issue if we misunderstood something or if there's something we missed out on or needs attention.

Thanks a lot for reporting this issue!

closed 2021-12-18 by MHShetty