GrapheneOS/Camera #80: The delay between the two consecutive flashes is too long on Pixel 3A

MHShetty, 2021-12-21:
We're currently waiting for the CameraX team to fix this issue (https://issuetracker.google.com/issues/211474332).

Quoting from the first message posted in that thread:

> We had previously filed an issue regarding the flash (https://issuetracker.google.com/issues/205373142) which has now been fixed with the latest release. However, our Pixel 3A users are facing another issue with the flash after the latest changes.
> 
> While clicking a photo there are two flashes (one to light the scene and another for the actual photo). Generally (for other devices), this delay between these two flashes is minimal (a few milliseconds).. however for Pixel 3a (specifically in the dark) this delay has been tested to be more than 1 second which indirectly causes the focus to get deviated while taking a photo. (The focus getting changed due to this was the main issue being faced by the user.

MHShetty, 2022-01-11:
This issue seems to have now been resolved by the CameraX team. Hope this issue gets resolved once the next release arrives and we migrate to it.

thestinger, 2022-01-26:
This should be fixed now.

closed 2022-01-26 by thestinger