GrapheneOS/Camera #86: Front-Cam video force close.

iaiaian1, 2022-01-04:
Hi, I have an issue where trying to do open front-cam video window force gives a force close.

This is my log. https://bin.kv2.dev/~61d38531410e7c32e6697918

This might be a device-related issue but still, I want to ask if you guys can take a look at this so I can fix it.

Tried versions: camera6 and latest camera8.

TLDR:
time: 1641252139929
--
  | msg: java.lang.IllegalArgumentException: Unable to find supported quality by QualitySelector
  | stacktrace: java.lang.IllegalArgumentException: Unable to find supported quality by QualitySelector
  | at androidx.camera.lifecycle.LifecycleCameraRepository.a(Unknown Source:13)
  | at androidx.camera.lifecycle.c.a(Unknown Source:42)
  | at h2.j.u(Unknown Source:66)
  | at l2.d.onClick(Unknown Source:25)
  | at android.view.View.performClick(View.java:7441)
  | at h2.e.onTouch(Unknown Source:3)

thestinger, 2022-01-04:
This is almost certainly a device-specific issue and is likely a device bug that CameraX would need to work around rather than something we can address in this app.

thestinger, 2022-01-07:
We need to know which device this happens on.

iaiaian1, 2022-01-09:
I am sorry for the late reply, was down with cold.

I am using a Redmi 7a/Pine running awaken OS. But no matter what rom the same bug happens. Note that gcam and stock camera works well.

Also, here is my local-manifest for building the awaken rom. Same trees are used every time.
https://github.com/iaiaian1/local-manifests/blob/Awaken-12.0/local_manifests.xml

thestinger, 2022-02-03:
Please test again on the most recent release.

iaiaian1, 2022-02-04:
I tested this 2/4/2022 right now. Front camera video still not working unfortunately.

Log:
https://katb.in/facewedakam

Zanthed, 2022-02-04:
> msg: java.lang.IllegalArgumentException: Unable to find supported quality by QualitySelector
> stacktrace: java.lang.IllegalArgumentException: Unable to find supported quality by QualitySelector

This seems relevant to #91 and needs more investigation if this is happening to devices outside of Android Studio emulator.

iaiaian1, 2022-02-13:
Version 15 log: https://bin.kv2.dev/~6209938b410e7c32e6699359

Zanthed, 2022-02-14:
The latest version didn't have any code changes. It was just updating dependencies and new icon.

thestinger, 2022-03-02:
@iaiaian1 Please try again with the latest version (version 13). CameraX made numerous improvements to video support for selecting resolution/bitrate.

Zanthed, 2022-03-05:
@iaiaian1 
Please test the latest release and see if the issue still persists. A fix has been made and should solve it for you.

Will close if there is no response.

iaiaian1, 2022-03-05:
Hi! I am still facing the same issue. Same log.

Does the latest version now works with other devices? If yes I can conclude that it is my device's issue and not on your side.

MHShetty, 2022-03-06:
Hi @iaiaian1,

Could you please test the latest release (v15) again?

iaiaian1, 2022-03-06:
V15 now works fine!

closed 2022-03-06 by iaiaian1