GrapheneOS/Camera #87: Zoom buttons like Google Camera

ccptr, 2022-01-05:
Google Camera has zoom buttons for 0.6x, 1x, and 2x. If you long press the buttons you can slide either left/right (or down/up in landscape) to zoom out/in. These buttons are always visible, allowing you to zoom with 1 finger.

thestinger, 2022-01-06:
Their zoom levels vary based on what the device supports. 0.6x, 1x and 2x is what they use on 5th generation Pixels where 0.6x is the wide angle camera without zoom, 1x is the main camera without zoom and 2x is 2x digital zoom with the main camera.

On the 6th generation devices, the main camera has a wider field of field so the wide angle camera button is 0.7x and there's a 4x telephoto camera too.

Adam-718, 2022-01-09:
In other words, it would be great it you can support all device cameras/lens while switching between these cameras made easy using the zoom slider.

thestinger, 2022-01-09:
We do support all the cameras *on Pixels* but you have to zoom normally to switch between them.

On most other devices, the alternate cameras are not hooked up for easy usage this way. It's unrealistic for us to hard-wire support for the cameras in other devices. We would need to hard-wire knowledge about the zoom levels they should be used on a device-specific basis. They should do it like Pixels and automate it in the OS instead of only hard-wiring it in their own camera apps.

Adam-718, 2022-01-12:
I appreciate your explaination so much!
Because I'm using this app on LineageOS I don't get the benefit of this multi lens support through the zoom slider, unfortunate.
Thanks again!

nsherbina1999, 2022-02-01:
> We do support all the cameras _on Pixels_ but you have to zoom normally to switch between them.
>
> On most other devices, the alternate cameras are not hooked up for easy usage this way. It's [...]

You can look up to PhotonCamera, how they implemented lens switches.

Donkey-Doug, 2022-03-25:
Can confirm that photon camera supports all the camera's on my xiaomi phone. 0.5x, 1x, 2x. Grapheneos only supports my 1x camera (while providing digital zoom)

thestinger, 2022-03-25:
Devices need to support the multi-camera API and CONTROL_ZOOM_RATIO in order to support automatically switching between the different cameras based on zoom level. Our camera app is not going to hard-wire the camera ids and zoom levels used by devices not implementing the Camera2 API for this functionality. It's up to your device vendor to provide a proper Camera2 implementation and it's entirely unrealistic for us to build a database of the cameras across all kinds of different phones none of the developers has access to. It's off-topic and has already been explained. Please don't bring it up on the issue tracker anymore. CameraX is where workarounds for broken/buggy/incomplete Camera API implementations is done, not inside our app. We aren't going to add device-specific quirks/workarounds here. CameraX is currently choosing to rely on device vendors implementing the appropriate Camera2 API rather than hard-wiring support for this for certain devices. It already works fine on Pixels and all other devices implementing the multi-camera API and CONTROL_ZOOM_RATIO properly.

thestinger, 2022-03-25:
@Adam-718 The multi-camera API and CONTROL_ZOOM_RATIO are standard parts of the Camera2 API and it's up to the device vendor and OS to implement them. Switching between cameras based on zoom level already works well on devices implementing these APIs. This issue is not about providing that.

thestinger, 2022-03-25:
This is not a camera app made for specific devices and the whole point of using CameraX is having a high level of portability with all the workarounds contained there, not inside the app. The appropriate place to discuss working around devices not implementing the Camera2 API and instead hard-wiring support for this feature in their vendor camera app is https://issuetracker.google.com/issues/221094921. We don't have access to those devices and fundamentally cannot maintain a hard-wired database of camera ids / zoom levels for hundreds of phones. It's entirely unrealistic.

thestinger, 2022-03-25:
The issue with this feature proposal is that we do not want to hard-wire the zoom levels for the cameras used by different devices. It's possible that the Camera2 multi-camera API can be used to retrieve the values, but currently very few devices implement this. Only devices implementing the multi-camera API and CONTROL_ZOOM_RATIO such as Pixels have support for automatically switching between zoom levels. It would defeat the purpose of that API if we hard-wired the zoom levels to provide buttons for certain devices. CameraX doesn't currently expose an API for retrieving this information so this cannot be implemented in a portable way, and we want our camera app to be portable instead of hard-wired to work fully with only certain devices.

ccptr, 2022-03-26:
The buttons wouldn't have to be hard wired for certain devices, but general buttons such as 0.5x, 1x, and 3x zoom.

thestinger, 2023-01-08:
Important to note that the 4x telephoto camera on the Pixel 6 Pro and Pixel 7 Pro is not actually triggered at 4x in many cases. It triggers at a dynamic zoom value based on the available light. Quite often, it doesn't switch until around 4.7x to 5x zoom and it may not trigger at all in very low light.