GrapheneOS/Camera #88: Custom storage location not working with a Nextcloud folder over DavX5

jel81, 2022-01-07:
The custom storage location is not working when set to a Nextcloud folder that's accessed with DavX5 (version 4.1-ose).
I can otherwise use the nextcloud folder and for example use the files app to copy photos into it. I can set the custom storage location to some local folder and that works. The camera app accepts the nextcloud folder in configuration but when I try to take a photo it says it can't access the folder.

thestinger, 2022-01-07:
Sounds more like a Nextcloud issue than a Camera app issue since it's using the standard API.

thestinger, 2022-01-07:
Try using a different storage provider.

jel81, 2022-01-07:
I can use the nextcloud folder otherwise, it's just gos camera app that has the issue

thestinger, 2022-01-07:
That doesn't imply that it's a GrapheneOS Camera app issue. Using it from the file manager is not using exactly the same APIs as in the file manager.

thestinger, 2022-01-07:
There isn't currently a reason to think that this is an issue with the GrapheneOS Camera app rather than the Nextcloud app's storage provider.

jel81, 2022-01-07:
> Using it from the file manager is not using exactly the same APIs as in the file manager.
Did you write that correctly? doesn't make any sense

thestinger, 2022-01-07:
Yes, I wrote it correctly and it does make sense. If Nextcloud's storage provider implementation doesn't work properly that isn't something we can address in this app. It's only an issue with this app if we're using SAF incorrectly. I don't think that's the case.

jel81, 2022-01-07:
I'm not a native speaker but a sentence that has 'file manger' both in the beginning and the end and 'not ... the same' in between seems to be self contradictory.

But I think I understand what you mean.

The different APIs that the camera app and the files app use, do they have a name or some other identifier?
Would you know of any other app that would use the same API as the camera app does?

jel81, 2022-01-07:
I now also tested with OpenCamera. It didn't have any problem in saving the photos in my nextcloud folder.

swimik, 2022-01-18:
I found this problem searching for another problem and think they might be related. I have nextcloud as well and when I take a picture with either the native graphene camera or another one(open camera, google camera) the picture gets uploaded to nextcloud but gets removed from the device after the upload is done.

I can post this as a separate issue if that is warranted but what was happening, was this:

Open camera app
Take picture
Picture would show up in the camera app thumbnail window like normal
A few moments later the picture would dissapear

If I take a picture and then quickly click on the thumbnail I could view it normally but then it would dissapear

If I take a picture and close the camera app, switch to gallery, I could see the photo briefly before it dissapeared as well.

In all three examples a blank .jpg would appear, and clicking on it would show two blank tiles, that appear to be two blank pictures written to the same filename.

I thought this was a permission error so I checked permissions and they seemed fine.
I reset all permissions and the camera, nextcloud, and gallery worked fine until grapheneos updated. Then it began again.

I don't know where to look at the log files on android to see which program is doing this, perhaps nextcloud is taking ownership of the photo before grapheneos does? And graphene is then blocking the write to disk because of the unprivlidged user file creation?

jel81, 2022-05-13:
An update, I tested today with the latest camera version and now it has started to work, sort of.

I still get an error but the photo is actually stored in the correct place. The exceptions looks like this

--- Exception 1 / 2
android.system.ErrnoException
fsync failed: EINVAL (Invalid argument)
fsync  ::  libcore.io.Linux  ::  -2
fsync  ::  libcore.io.ForwardingOs  ::  274
fsync  ::  libcore.io.BlockGuardOs  ::  205
fsync  ::  libcore.io.ForwardingOs  ::  274
fsync  ::  android.system.Os  ::  204
Q  ::  l2.e  ::  34
run  ::  l2.d  ::  1
runWorker  ::  java.util.concurrent.ThreadPoolExecutor  ::  1167
run  ::  java.util.concurrent.ThreadPoolExecutor$Worker  ::  641
run  ::  java.lang.Thread  ::  920
--- Exception 2 / 2
l2.f
android.system.ErrnoException: fsync failed: EINVAL (Invalid argument)
Q  ::  l2.e  ::  38
run  ::  l2.d  ::  1
runWorker  ::  java.util.concurrent.ThreadPoolExecutor  ::  1167
run  ::  java.util.concurrent.ThreadPoolExecutor$Worker  ::  641
run  ::  java.lang.Thread  ::  920

thestinger, 2022-05-14:
Should work with version 37.

closed 2022-05-14 by thestinger