GrapheneOS/Camera #89: 3:2 aspect ratio

flawedworld, 2022-01-08:
This was requested by a user.

MHShetty, 2022-01-10:
Hey @flawedworld,

We'll need to wait for CameraX to support it and to be able to support 3:2 aspect ratio. (We have filed an feature request for the same that can be tracked from here: https://issuetracker.google.com/issues/213757975.)

Thanks a lot for filing this feature request!

MHShetty, 2022-01-10:
They're asking for more information regarding the same (like what are the use cases for 3:2 etc.). 

Could you please ask the user about the same?

flawedworld, 2022-01-10:
3:2 aspect ratio is a standard aspect ratio used in the majority of standalone digital cameras.

MHShetty, 2022-01-11:
Hi @flawedworld,

I have mentioned this use-case there and there are a few more new messages there. It might take a while before this gets implemented, but it seems that they have accepted our feature request.

Thanks a lot for filing this feature request!

lbdroid, 2022-01-12:
I don't see how this could actually be "blocked on CameraX". The native sensor resolution is going to be 4:3, therefore to achieve 3:2 aspect ratio, or for that matter ANY other aspect ratio, would involve cropping the preview and output images. While the HAL and/or CameraX would be responsible for cropping normally, it is something that can easily be done by the application as well.

thestinger, 2022-05-19:
Seems to too niche.

closed 2022-05-19 by thestinger

thestinger, 2022-05-23:
Image capture cropping can already be done with the available CameraX APIs. Video capture doesn't have the necessary APIs yet. I don't really see much point in a purely software implementation of 3:2. The advantage of being able to use hardware support would be better performance. The native image sensor resolution isn't necessarily 4:3 but I'm not sure how common other aspect ratios would be. Video oriented devices may have a wider native aspect ratio but it's likely going to be wider than 3:2. 3:2 seems quite niche and I'm unsure why it's being requested specifically. We can consider supporting other aspect ratios in the future but for the time being I see little reason to add support for arbitrary aspect ratios without a way to use hardware support.

thestinger, 2022-05-23:
It also needs to be clarified whether they're requesting it for image capture, video capture or both of them. It's not at all the same.