GrapheneOS/Camera #90: Some pictures in the gallery are accessible from the lockscreen

Kruptys, 2022-01-10:
Hello, I think I found a bug in the app that allows you to access, from the lockscreen, some pictures stored in the gallery.
Of course, you must have the "Screen lock camera access" option enabled.
As I don't think this is an expected behavior, that's why I opened this issue.

The pictures accessible from the lockscreen seem to be only those taken with a recent version of the app, as I can't access pictures taken from early builds of the app or the previous AOSP camera app.

Here are the steps to reproduce the bug :
- Lock your screen by pressing the power button
- Double press the power button to launch the camera app from the lockscreen
- Take a picture
- Click on the gallery button
- Delete your picture
- You are now back on the camera screen
- Click again on the gallery button
- There you go, you have access to recent pictures stored in the gallery of the profile

Camera app version : 8
Phone : Pixel 4a

thestinger, 2022-01-10:
This was a regression caused by a recent bug fix. It should be fixed by https://github.com/GrapheneOS/Camera/commit/1d714d56082f1d913654b327d76347b24d0fa2f2.

closed 2022-01-10 by thestinger

thestinger, 2022-01-10:
There's a new release (version 9) with the fix. It will be bundled into a new release of GrapheneOS soon.