GrapheneOS/Camera #91: App crash when unable to detect supported video quality type

Zanthed, 2022-01-15:
This is just a general edge-case bug I found during some random testing in the Android Studio emulator. App will crash if it can't detect a supported video quality type when switching to Video recording.

This is not a bug related to GrapheneOS, just a general edge-case bug.

```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: app.grapheneos.camera, PID: 2262
    java.lang.RuntimeException: Unable to resume activity {app.grapheneos.camera/app.grapheneos.camera.ui.activities.MainActivity}: java.lang.IllegalArgumentException: Unable to find supported quality by QualitySelector
        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4797)
        at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4830)
        at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:54)
        at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
        at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2252)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loopOnce(Looper.java:201)
        at android.os.Looper.loop(Looper.java:288)
        at android.app.ActivityThread.main(ActivityThread.java:7842)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
     Caused by: java.lang.IllegalArgumentException: Unable to find supported quality by QualitySelector
        at androidx.camera.lifecycle.LifecycleCameraRepository.bindToLifecycleCamera(LifecycleCameraRepository.java:280)
        at androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:532)
        at androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:381)
        at app.grapheneos.camera.CamConfig.startCamera(CamConfig.kt:1244)
        at app.grapheneos.camera.CamConfig.initializeCamera(CamConfig.kt:1101)
        at app.grapheneos.camera.ui.activities.MainActivity.onResume(MainActivity.kt:519)
        at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1488)
        at android.app.Activity.performResume(Activity.java:8194)
        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4787)
        at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4830) 
        at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:54) 
        at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45) 
        at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176) 
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2252) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loopOnce(Looper.java:201) 
        at android.os.Looper.loop(Looper.java:288) 
        at android.app.ActivityThread.main(ActivityThread.java:7842) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003) 
```

thestinger, 2022-01-26:
Needs to be tested with the latest release.

thestinger, 2022-01-26:
Needs to be tested with the latest beta release.

Zanthed, 2022-01-26:
This is still an issue with the newest beta.

Zanthed, 2022-01-26:
This only happens with obscure camera configs like emulator, but something in the app's code in CamConfig.kt is doing something wrong.

Zanthed, 2022-03-05:
Solved by latest release.

closed 2022-03-05 by Zanthed