GrapheneOS/Camera #103: QR Scanner crash

nsherbina1999, 2022-02-01:
```
E/MessageQueue-JNI: java.lang.IllegalArgumentException: Invalid extension mode!
        at androidx.camera.extensions.ExtensionsInfo.getExtendedCameraConfigProviderId(ExtensionsInfo.java:287)
        at androidx.camera.extensions.ExtensionsInfo.getFilter(ExtensionsInfo.java:195)
        at androidx.camera.extensions.ExtensionsInfo.isExtensionAvailable(ExtensionsInfo.java:134)
        at androidx.camera.extensions.ExtensionsManager.isExtensionAvailable(ExtensionsManager.java:413)
        at app.grapheneos.camera.CamConfig.startCamera(CamConfig.kt:1164)
        at app.grapheneos.camera.CamConfig.switchMode(CamConfig.kt:1500)
        at app.grapheneos.camera.ui.activities.MainActivity.finalizeMode(MainActivity.kt:1049)
        at app.grapheneos.camera.CamConfig.loadTabs$lambda-15$lambda-14$lambda-13$lambda-12(CamConfig.kt:1379)
        at app.grapheneos.camera.CamConfig.$r8$lambda$Ds0vyOYJx4eUBiFjSYJaqoFjx2E(Unknown Source:0)
        at app.grapheneos.camera.CamConfig$$ExternalSyntheticLambda5.onTouch(Unknown Source:4)
        at android.view.View.dispatchTouchEvent(View.java:14372)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3112)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2785)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3118)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2799)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3118)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2799)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3118)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2799)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3118)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2799)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3118)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2799)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3118)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2799)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3118)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2799)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3118)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2799)
        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3118)
        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2799)
        at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:522)
        at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1877)
        at android.app.Activity.dispatchTouchEvent(Activity.java:4193)
        at androidx.appcompat.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:69)
        at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:480)
        at android.view.View.dispatchPointerEvent(View.java:14635)
        at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:6166)
        at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:5969)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:5460)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5517)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:5483)
        at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:5635)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:5491)
        at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:5692)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:5464)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5517)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:5483)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:5491)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:5464)
        at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:8309)
E/MessageQueue-JNI:     at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:8225)
        at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:8186)
        at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:8445)
        at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:227)
        at android.os.MessageQueue.nativePollOnce(Native Method)
        at android.os.MessageQueue.next(MessageQueue.java:342)
        at android.os.Looper.loop(Looper.java:197)
        at android.app.ActivityThread.main(ActivityThread.java:7755)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:958)
D/AndroidRuntime: Shutting down VM
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: app.grapheneos.camera, PID: 21920
    java.lang.IllegalArgumentException: Invalid extension mode!
        at androidx.camera.extensions.ExtensionsInfo.getExtendedCameraConfigProviderId(ExtensionsInfo.java:287)
        at androidx.camera.extensions.ExtensionsInfo.getFilter(ExtensionsInfo.java:195)
[... 36 lines cut ...]
        at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:5969)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:5460)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5517)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:5483)
        at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:5635)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:5491)
        at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:5692)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:5464)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5517)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:5483)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:5491)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:5464)
E/AndroidRuntime:     at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:8309)
        at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:8225)
        at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:8186)
        at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:8445)
        at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:227)
        at android.os.MessageQueue.nativePollOnce(Native Method)
        at android.os.MessageQueue.next(MessageQueue.java:342)
        at android.os.Looper.loop(Looper.java:197)
        at android.app.ActivityThread.main(ActivityThread.java:7755)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:958)
```

Zanthed, 2022-02-01:
What device is this?

nsherbina1999, 2022-02-02:
Motorola Moto G8 Power, Android 11, stock AOSP with Moto Actions.

nsherbina1999, 2022-02-02:
Also, sometimes there are additional modules (HDR, for example), and sometimes there is no any of them, nor in frontal, nor in back camera.
If there are additional modules, QR Scanner won't work, but if there is no any, it's working as intended.

thestinger, 2022-02-02:
Do the camera and video modes still work properly? Do the other modes work?

nsherbina1999, 2022-02-02:
Yes, video and camera works OK. But when HDR and another extensions is exists in modes, QR Scanner is crashing. I don't know how it works, sometimes I have modes, sometimes not.

thestinger, 2022-02-02:
@nsherbina1999 Please try this beta build with a potential fix for the crash:

https://github.com/GrapheneOS/platform_external_Camera/raw/12/prebuilt/Camera.apk

nsherbina1999, 2022-02-02:
Now it works good. Thanks! Also, I've found out that QR Scanner at frontal camera returns on back camera. If that was the intent, it could be confusing to people, and I think it would be easier to remove that clause.

thestinger, 2022-02-02:
@nsherbina1999 Which modes appear for you have other than QR scanner, Camera and Video? Can you test the HDR mode and try to see if it actually appears different than the Camera mode? We intended to have support for vendor extensions which can include Portrait, HDR, Night Sight, Face Retouch and Auto but we haven't been able to test it ourselves. That's why this bug existed. Pixels haven't shipped them yet.

thestinger, 2022-02-02:
Ideally we can fix whatever causes the vendor extension modes to only appear for you some of the time but it's possible that it's a bug in CameraX or the vendor's extension implementation. I don't see anything wrong with our code...

closed 2022-02-02 by thestinger

thestinger, 2022-02-03:
There's a new release (v11) with this change incorporated. Would appreciate if you could try testing the vendor extensions more and if we get a better understanding of what's going wrong we can potentially fix it. It seems like there must be some race condition with the asynchronous initialization but I really can't see any issues with our code.

nsherbina1999, 2022-02-03:
Yes, it works good now. Thanks! 
Also, in first launch I don't have any extensions, but after reloading an app, it works good. IDK, maybe it's CameraX bug... Screenshots attached.

In first launch:
[image]
[image]

After app reload (removing from recent and opening it again):
[image]
[image]

thestinger, 2022-02-03:
By first launch do you mean after switching away from the app and reopening, or do you need to close the activity? Can you try both?

thestinger, 2022-02-03:
Force stopping it would actually be the best way to test. If it works fine after force stopping it then it must be an OS issue. If the same problem occurs after force stop, it could be an OS issue, but it's also possible that it's a CameraX bug or a bug in our code. Not sure which is most likely at this point.

nsherbina1999, 2022-02-10:
After force stop it's ok. But in first launch I don't have any extensions.