GrapheneOS/Camera #104: very laggy options animation

steve47457, 2022-02-01:
[image]
Please fix

thestinger, 2022-02-01:
Unable to reproduce this. You'll need to give more details on the phone. It may be too slow to do animations smoothly when using the camera.

closed 2022-02-01 by thestinger

Zanthed, 2022-02-01:
The Android Studio Emulator is not a good reference as it's not great in real-world performance. It's meant for basic testing and development.

Please use a real device and see if the issue persists.

steve47457, 2022-02-02:
@Zanthed I have the same issue on my phone. I just used the emulator for demonstration purposes

steve47457, 2022-02-02:
@thestinger  Samsung Galaxy A03s

thestinger, 2022-02-02:
I think it's simply an issue with using the camera on low-end phones. They're struggling to simply support the modern Camera2 API including the preview and can't do smooth animations at the same time. I don't think we can do anything about this. We're just using a standard animation library.