GrapheneOS/Camera #108: Incorrect layout on Samsung Galaxy A03s: large gap at bottom of screen

steve47457, 2022-02-02:
Incorrect layout on Samsung Galaxy A03s: large gap at bottom of screen:
[image]

MHShetty, 2022-02-04:
Hey @steve47457,

Are you able to consistently reproduce this layout/behavior on your device?

programmer57, 2022-02-08:
@MHShetty  Yes, this is how it always looks like on my device.

steve47457, 2022-02-13:
@MHShetty Yes, it has always been like this.

steve47457, 2022-02-13:
Are there plans to fix this in future updates? @MHShetty

MHShetty, 2022-06-03:
Hi @steve47457,

We could try fixing this issue on our end (unless it isn't being caused due to unexpected behavior from the OS). Could you please share an screenshot or describe how the app looks when the aspect ratio is set to 16:9 mode?

Extremely sorry for the late reply. Thanks a lot for reporting this issue @steve47457!

MHShetty, 2022-06-03:
Hi @steve47457,

We could try fixing this issue on our end (unless it isn't being caused due to unexpected behavior from the OS). Could you please share an screenshot or describe how the app looks when the aspect ratio is set to 16:9 mode?

Really sorry for the late reply and thanks a lot for reporting this issue @steve47457!

r3g-5z, 2023-02-23:
Is this still an issue?