GrapheneOS/Camera #109: Request

007chaand, 2022-02-03:
Please add support RAW and manual mode 🙏

thestinger, 2022-02-03:
See https://github.com/GrapheneOS/Camera/issues/14 for RAW support. You'll need to clarify which features you want from a manual/professional mode since they would need to be separately implemented.

closed 2022-02-03 by thestinger

007chaand, 2022-02-23:
> See #14 for RAW support. You'll need to clarify which features you want from a manual/professional mode since they would need to be separately implemented.

I mean full manual control:
Manual focus
Manual iso
Manual exposure
Manual white balance
Raw/DNG support

thestinger, 2022-02-23:
It needs to be requesting separated from RAW support and what's being requested needs to be more specific.

locustangerine, 2022-03-08:
What's the status on creating a feature request for a new camera mode that has 007chaand's requested features? A good reference is the stock camera app for the OnePlus 6: sliders and wheels that allow Auto or Manual settings for focus, white balance, shutter speed, ISO, etc.

[image]

007chaand, 2022-03-09:
Yes that exactly what we want 😊
By the way I love graphene camera app