GrapheneOS/Camera #111: [Feature request] allow third-party gallery apps

BluePixel4k, 2022-02-04:
Please allow third-party gallery apps. 
I like to use the simple gallery app to display the images and so that I have a unified recycle bin.

Zanthed, 2022-02-04:
You already can use third-party gallery apps just fine. Need to clarify what you're asking.

Zanthed, 2022-02-04:
Are you asking about using a third-party gallery app instead of the built-in app gallery?

BluePixel4k, 2022-02-04:
Yes exactly :)

thestinger, 2022-06-04:
This can't realistically be supported due to SAF, latency required to fetch all of the URLs to pass to the app, etc. Our in-app gallery is not simple showing all photos in a specific directory. It's showing the photos taken by the app for that storage location tracked by MediaStore / SAF.

closed 2022-06-04 by thestinger