GrapheneOS/Camera #117: Feature Request: Need more camera features

UnixPhonez, 2022-02-17:
Some camera features needed

- Night mode
- Portrait mode
- Long exposure mode 
- Time lapse mode
- Slow mode
- Pano mode
- Ability to set filters via a swipeable tab 
- Ability to set filters for portrait mode such as natural light, studio, contour, stage light, stage light mono, and high key light mono

Zanthed, 2022-02-17:
Majority of these are blocked until Google releases these CameraX extensions for us to use and the necessary APIs provided by AndroidX CameraX.

thestinger, 2022-02-17:
There are no plans to implement filters since it can be done externally and is outside the scope of the app.

The rest of these depend on CameraX vendor extensions and most already have issues filed. A feature request should only request 1 specific thing.

closed 2022-02-17 by thestinger

thestinger, 2022-02-26:
Night mode
Portrait mode

These are already supported on a device with the CameraX vendor extensions for them by the way.