GrapheneOS/Camera #121: [Feature Request] Add option to manually switch between lenses for Pixel 6 Pro

frostelk, 2022-02-25:
The zoom slider switches from the normal lens to the telephoto lens automatically as you zoom in, but the level of magnification that causes it to switch seems to vary between around 4.5 and 6x. I would like the option to select the telephoto camera so I can take photos using only optical zoom and capture the maximum resolution; right now taking photos at 4x zoom automatically uses the standard lens and not telephoto.

Zanthed, 2022-02-26:
duplicate of https://github.com/GrapheneOS/Camera/issues/87

closed 2022-02-26 by Zanthed

thestinger, 2022-02-26:
You can zoom to the point that it switches and then go back.

frostelk, 2022-03-01:
> duplicate of #87

This is similar to #87 but is not a duplicate because the 4x zoom button in Google Camera has the same issue in that it does not switch to the telephoto lens unless the user zooms in further. Anecdotally, it seems that the magnification "threshold" that triggers the lens switch is higher in Google Camera (~7-9x) than it is in GrapheneOS Camera (~4.5-6x) after experimenting with them side-by-side. If the button-zoom approach described in #87 is implemented, it could also address this issue if the 4x zoom button acted as a strict lens switcher.

> You can zoom to the point that it switches and then go back.

This does not work reliably when I have tried it. Usually it is a slight improvement -- if the lens in use switches to telephoto zooming in at 5.5x zoom, it will stay in use as I zoom out until 5x zoom, as an example -- but does not offer the full functionality I am describing.

thestinger, 2022-03-01:
I think this is only an issue on the Pixel 6 and will probably get fixed.

thestinger, 2022-03-01:
It would be helpful to know if it happens in the stock OS.

frostelk, 2022-03-01:
> I think this is only an issue on the Pixel 6 and will probably get fixed.

Great thank you for your response!

thestinger, 2022-03-25:
This is an upstream bug with the Pixel 6 also impacting Google Camera itself so we're simply going to need to wait for it to be resolved for the OS / firmware camera support.