GrapheneOS/Camera #122: [bug] app is not using all the screen and hides part of the settings button

williamdes, 2022-02-26:
[image]

Here is the screenshot of exactly what I see.
I am using a Redmi Note 10 Pro. So if you do not do support for other devices you can close this issue.

The app should someway understand that I have a large screen and not render some kind of black overlay that hide the settings button

thestinger, 2022-02-26:
> So if you do not do support for other devices you can close this issue.

We do want to support other devices. We just don't want to have to write device-specific code and we're leaving that up to CameraX. We can work on addressing this issue.

empratyush, 2022-03-05:
hey @williamdes  i think you are having this issue because you are using "hide notch" feature from MIUI which works in non-standard way.

there is standard api for checking available screen real state (for third party devs) but somehow MIUI is breaking standard api to provide "hide notch" feature. Try tunring "hide notch" feature off.

williamdes, 2022-03-05:
> hey @williamdes i think you are having this issue because you are using "hide notch" feature from MIUI which works in non-standard way.
> 
> there is standard api for checking available screen real state (for third party devs) but somehow MIUI is breaking standard api to provide "hide notch" feature. Try tunring "hide notch" feature off.
> 
> edit: [MIUI have same issue in other mi devices too ](https://drive.google.com/file/d/1DLH8p0VQdyFwFnGhQ9MxBEVkK_QXavM7/view?usp=sharing)

You are right, that said I do no like the notch thing so for now disabling will not be a solution
But thank you for feed back this is interesting. I hope you can find a fix someway. If not that not to much of a problem as I can click the menu button and there is nothing more to see behind.

empratyush, 2022-03-06:
>  I hope you can find a fix someway. 

We don't want to write device-specific code when oem deliberately break standard api compatibility . standard api exist for a reason.

closed 2022-03-06 by empratyush

williamdes, 2022-03-26:
Just updated to Android 12/MIUI 13 and it is fixed! 

[image]