GrapheneOS/Camera #123: [Feature request] front camera "flash"

BluePixel4k, 2022-02-27:
In other camera apps there is a "flash" for the front camera mode, which is realized by brightening the screen and changing the GUI color to creamy white.

Would be great if that can be added to this app too, thanks.

GCam example:
[image]

Zanthed, 2022-02-27:
Already exists. It's called "Self-Illumination".

closed 2022-02-27 by Zanthed

BluePixel4k, 2022-02-28:
Oh thank you, I overlooked that. Would it be conceivable that you add the function to the flash function, instead of the separate checkbox? That would be more intuitive from my point of view.

Currently there is a message that the flash is is unavailable in the current mode.

[image]