GrapheneOS/Camera #124: [Feature request] F-Droid / IzzyOnDroid Repo

BluePixel4k, 2022-02-27:
It would be handy if the camera app can also be updated via the F-Droid store.
The PFD Viewer and Auditor App are already in the IzzyOnDroid repo. (Strangely, I can't see the Auditor app in F-Droid anymore, recently it was there?).

Zanthed, 2022-02-27:
We are not going to package anything on F-Droid due to the recent hostility, misinformation, and attacks on the project by F-Droid core developers and their supporters along with the security issues it has and technical issues it has.

We have taken down Auditor and anyone else packaging our apps on F-Droid is going to be asked to take it down. This includes PDFViewer which needs to be taken down. Izzy is not allowed to package our apps anymore and no one else either.

closed 2022-02-27 by Zanthed

Zanthed, 2022-02-27:
If you'd like an app store client to obtain our software from that is actually secure and built with proper UX in mind, we have an Apps client which can be obtained here for now: https://github.com/GrapheneOS/platform_external_Apps/blob/12/prebuilt/Apps.apk

thestinger, 2022-02-27:
We're replacing F-Droid for technical reasons to provide reasonable usability and security. F-Droid developers took offense to that and are now involved in spreading misinformation about GrapheneOS and severe harassment. They coordinated a raid on our Auditor issue tracker. F-Droid is a thoroughly untrustworthy project. It isn't going to be supported in any way and  they aren't welcome to package our apps. One of their core developers is involved in substantial efforts to direct harassment towards me across multiple Telegram/Matrix rooms and Twitter over the past 24 hours.