GrapheneOS/Camera #125: Feature Request: Save Images as AVIF (as default)

heuwerk, 2022-02-28:
Is it possible to add AVIF support in the camera app or maybe even use it as default?

thestinger, 2022-02-28:
It has to be supported by the device or there's no point since it would involve conversion from JPEG. Most camera apps supporting saving in alternate formats are really just converting from JPEG. That doesn't really make sense.

RAW support would be required to convert to arbitrary formats properly. We can consider supporting other formats once https://github.com/GrapheneOS/Camera/issues/14 is implemented.

closed 2022-02-28 by thestinger

BUAWF, 2022-05-11:
> It has to be supported by the device or there's no point since it would involve conversion from JPEG. Most camera apps supporting saving in alternate formats are really just converting from JPEG.

@thestinger  
Hi there,

I'm using Samsung Galaxy S10 Lite (SM-G770F) Snapdragon 855 here in Qatar.  I've been looking at Your work since the days of CopperHeadOS. I Adore Your Work.

Default Samsung One UI Camera App, I (at the very first) change the Image to Heic & the Video to Hevc.  Is this the same as You have mentioned here that, it's just a conversion from JPEG to HEIC/HEIF for Images & from AVC to HEVC for Videos  in My Device too ?

My Phone is Completely Stock (One Ui 4.0) no root nothing (using it as a normal user).

My Settings for the UI are No Battery Optimization (no App is Optimised, ALL are Unrestricted), No Personalisation from Google (even Usage & Diagnostics, Android Personalisation Service is also off), No Customisation Service, Send Diagnostic Data (by Samsung Account is also turned off).
Screen Font size is completely to the smallest + Screen Zoom to entirely zoomed out,  Font, I've changed from Default to SamsungOne, even in the Samsung Internet Browser in Settings --> Labs --> Font is SamsungOne.