GrapheneOS/Camera #126: Unable to launch camera from lockscreen by double-tapping power button on Pixel Pro 6

frostelk, 2022-03-01:
I have enabled lockscreen camera and quick camera access via:

Settings -> Security -> Screen lock camera access
Settings -> System -> Gestures -> Quickly open camera

Double-tapping the power button successfully opens the camera when the device is unlocked, though there is a significant lag and it looks as though the device locks, unlocks, and then opens the camera app. When the device is locked, double-tapping the power button prompts for the password and I am unable to access the camera.

Zanthed, 2022-03-01:
Do you have multiple camera apps installed?

frostelk, 2022-03-03:
I have Google Camera installed in a work profile managed with Shelter -- since it is in Shelter it cannot be set as the default app for double-tap launch. I reset the default app links and just to be sure I am preventing Google Camera from opening links at all, but GrapheneOS Camera still will not launch from the lockscreen. Is there any way that I can access the camera from the lockscreen other than with the double-tap?

thestinger, 2022-03-03:
> Is there any way that I can access the camera from the lockscreen other than with the double-tap?

No, since the swipe feature for that is no longer enabled in AOSP. We don't plan to enable it, and I'm not sure if it's even still around.

frostelk, 2022-03-03:
> > Is there any way that I can access the camera from the lockscreen other than with the double-tap?
> 
> No, since the swipe feature for that is no longer enabled in AOSP. We don't plan to enable it, and I'm not sure if it's even still around.

Damn that's too bad. Would you be able to direct me to more information on resetting app link associations "in a very specific way?" Thanks for your help!

thestinger, 2022-03-03:
I don't have a link to specific information about it. It's an AOSP bug introduced in Android 12 which impacts the stock OS too. It's having multiple camera apps which triggers it. You should try doing it a few different ways in a different order like trying to set the lockscreen intent default first before the non-lockscreen one. We don't have the exact instructions on working around it.

frostelk, 2022-03-03:
Alright, thank you for your response. I will update here if I find a reliable procedure

e-t-l, 2023-01-22:
I'm also encountering this issue, and I have Open Camera and nothing else. The Graphene Camera app is Disabled, because I don't see a way to uninstall it completely on the main profile (although it *can* be uninstalled from the work profile). I've done the suggested fixes, including resetting the camera default app link associations. Is there any other factor that might be affecting this?

thestinger, 2023-01-22:
> The Graphene Camera app is Disabled

Only system camera apps can provide the system media capture intents since Android 11 so disabling this is broken and will break apps using those. We'll have to prevent doing this.

thestinger, 2023-01-22:
> Is there any other factor that might be affecting this?

A bug in Open Camera.

closed 2023-01-23 by thestinger

e-t-l, 2023-01-23:
> Only system camera apps can provide the system media capture intents

In what way? I haven't noticed any capture-related features no working.

> A bug in Open Camera

That's a nice thought. Unfortunately it's true regardless of what camera app is installed, including Gcam, and even if it is the only one enabled.

thestinger, 2023-01-23:
> In what way? I haven't noticed any capture-related features no working.

I explained. You can research if you want to know more. It was a widely announced and covered Android 11 change.

> That's a nice thought. Unfortunately it's true regardless of what camera app is installed, including Gcam, and even if it is the only one enabled.

Nope.