GrapheneOS/Camera #129: Upgrade UI style and components to Material You

jorgeluiscarrillo, 2022-03-04:
Currently we are hardcoding some colors using the GrapheneOS system palette. We should migrate to the modern Android design style supporting dynamic color. Additionally, any components (buttons, dialogs, preferences etc.) should also use the new style.

harishkumarankr2004, 2022-11-25:
When will this be implemented fully? 
Will it look similar to Google Camera?
If yes, that's great news since many people who don't have pixels will also have access to a camera app that is of great quality and ux too.

NXTler, 2023-09-23:
I add my vote to this

thestinger, 2023-10-06:
Please use reactions instead of comments for +1.

MHShetty, 2025-02-13:
Hi @jorgeluiscarrillo,

This has been addressed in a PR that was merged a while ago. Please do let us know if you find anything that might have accidentally been left out by any little chance.

Thanks a lot for trying our camera app and filing this issue!

closed 2025-02-13 by MHShetty