GrapheneOS/Camera #131: Settings button hidden under the front camera

VHUGO98, 2022-03-04:
[image]

The button is getting hidden under the front camera, which ends up making it difficult to access through it 

Device info:

Motorola Moto G9 Plus (Odessa)
Android 11
Qualcomm Snapdragon 730G
1080 x 2400 pixels, 20:9 ratio 
6.81 inches

thestinger, 2022-03-26:
This is likely due to your vendor not properly implementing the API for handling display cutouts. Do you have any non-default settings or developer options enabled changing how that works? We've seen this issue with certain other devices. Xiaomi appears to have fixed it for their Android 12 release.

thestinger, 2022-04-12:
This appears to be a Motorola bug. Not much we can do.

closed 2022-04-12 by thestinger

ghostrider-reborn, 2022-08-22:
> This appears to be a Motorola bug. Not much we can do.

Currently facing this issue on a custom AOSP 13 build on Realme X3. The cutout path in my overlay is defined:

```
     <string name="config_mainBuiltInDisplayCutout">
        M -544,0 L -544,108 L -270,108 L -270,0 Z @left
     </string>
```

Any ideas?

thestinger, 2022-08-22:
@ghostrider-reborn It must not be set up correctly by your OS. There's probably more to it than that. We don't know the setup process, just how to use it from an app.

ghostrider-reborn, 2022-08-22:
> @ghostrider-reborn It must not be set up correctly by your OS. There's probably more to it than that. We don't know the setup process, just how to use it from an app.

Got it, the cutout path was to blame. I recreated it manually and now it works fine.

```
     <string name="config_mainBuiltInDisplayCutout">
        M 0,0 H 254 V 108 H 0 V 0 Z @left
     </string>
```