GrapheneOS/Camera #134: App size is smaller

cyberpunkme, 2022-03-06:
Can u provide another direct link for me 
[Look at this ](https://telegra.ph/Is-this-only-for-me-03-06)

thestinger, 2022-03-06:
That's the correct size, The units are different.

thestinger, 2022-03-06:
You can install this app from the Play Store or from our app repository. The one in our app repository is the same as the one published on GitHub.

https://github.com/GrapheneOS/Apps/releases/tag/4

Our app repository will update itself and this app once you install both.

cyberpunkme, 2022-03-06:
[image]

thestinger, 2022-03-06:
The app requires Android 10+.

cyberpunkme, 2022-03-06:
Heyy brother please provide appcenter link

closed 2022-03-06 by thestinger

cyberpunkme, 2022-03-06:
> The app requires Android 10+.

Ohh ok bye

thestinger, 2022-03-06:
There's a variant of the app on the Play Store and the standalone variant in our app repository. The standalone variant is what's published on GitHub.