GrapheneOS/Camera #135: Support for Older Android Versions

moriel5, 2022-03-06:
As I can see, the CameraX library was designed to work with as far back as Android 5, however Secure Camera requires at least Android 10, which prevents me from testing it with my devices, running Android 7.1.2 (Nexus 4, LG G3) and 9 (Razer Phone 2) respectively.

How feasible would it be to add support for older versions, should it make sense to do so?

thestinger, 2022-03-06:
We're not planning on supporting anything older than Android 10 for Camera. It uses MediaStore APIs which were introduced in Android 10 and we don't want to use legacy storage.

Android 9 is the oldest version with security support and will be the new minimum across our apps soon.

thestinger, 2022-03-06:
We're already having a lot of trouble supporting devices with poor Camera2 API support.

closed 2022-03-06 by thestinger

moriel5, 2022-03-06:
That is certainly understandable.

Thanks for the answer, I hope that I'll manage to install a custom ROM on my Razer Phone 2 as soon as possible (I have been doing this for years, however for some reason, since Android 8, I keep getting bootloops on custom ROMs, regardless of the device, with no apparent rhyme or reason (just randomly, it could start after 1 day, 1 week, or anything in between and beyond), except for my Razer Phone 2, where I keep getting bootloops immediately upon flashing custom ROMs ( TWRP work fine, though)).

dumblob, 2022-10-14:
@moriel5 did you find a way to solve your issues? I am in the very much same situation.

Actually I am considering forking this repo just to compile the camera for older API (my target would be API 25) while disabling (instead of fixing/patching/shimming/polyfilling) all incompatible functionality (even if it would make the camera not useful - I am interested in testing the camera behavior first; not trying to use it to actually save pictures).

What do you think?

Admittedly my primary motivation is the "zero shutter lag" feature of Secure Camera from GrapheneOS. But maybe that particular functionality is not available on older devices (or maybe on no devices except for Pixels; IDK).

moriel5, 2022-10-15:
@dumblob To say the truth, I haven't yet found the time to try replacing the ROM again (too many real-world things to take care of, many of which are also related to tech, such as rewiring network cables in the walls of our house).

If you do fork Secure Camera, I would love to test it out, as I haven't yet found the time to understand how to compile Android apps (I'm used to compiling with CMake, Ninja and Make, especially when packaging software for Solus, but not with whatever is required for Android).

Regarding features, that shall be seen.

thestinger, 2022-10-15:
You just need to run `./gradlew assembleRelease` and you need OpenJDK installed for Gradle to work. You may also need the Android SDK set up, but I haven't tried without it and it may be able to fetch that itself. Gradle will fetch the dependencies including fetching a compatible OpenJDK toolchain if you don't have one. You then need to sign the release if you didn't create a keystore and configure it in advance before running assembleRelease (assembleRelease signs the build itself if that's configured locally). If you do `./gradlew assembleDebug` that will always automatically sign with your global Android SDK debug key generated the first time you start doing builds.

moriel5, 2022-10-16:
@thestinger Thanks, that is very helpful.

If I'll have time, I'll try doing that. At least OpenJDK is officially packages for Solus, and I believe that I can unofficially package Android Studio (or perhaps just the SDK if it isn't already packaged, we already have ADB and Fastboot packaged), plus I already have a GPG key set up for signing.

dumblob, 2022-10-17:
I am also very busy but maybe I will get to it - thanks @thestinger for building instructions, appreciated!