GrapheneOS/Camera #136: When switching to Video mode . App crashes. (system-ColorOS12.1)

ChAJ07, 2022-03-06:


ChAJ07, 2022-03-06:
https://user-images.githubusercontent.com/101049867/156925104-9399b077-d1fb-4831-b4dd-d6f3c41b5ee0.mp4

Device- realme 8 4G(RMX3085)
Processor- Helio G95
Android- 12
ColorOS ver.- 12.1

Zanthed, 2022-03-06:
Please make sure you're testing this on the newest version (16).
https://github.com/GrapheneOS/Camera/releases/tag/16

If you obtained it from Play Store, you will need to wait a couple days for version 16 to show up.

If the issue still persists on version 16, please post logs from logcat.

thestinger, 2022-03-06:
@ChAJ07 Several issues were worked around in the latest release (version 16). Play Store still only has version 15 because the new version is still in the review process. It would be very helpful if you could uninstall the Play Store release and try using the one we publish on GitHub and in our app repository instead. You can switch back to the Play Store version once version 16 passes review, which will hopefully be later today or early tomorrow.

ChAJ07, 2022-03-07:
I'm on ver. 16 (latest from github). Bug isn't fixed. I'm sharing logs @Zanthed

ChAJ07, 2022-03-07:
https://drive.google.com/file/d/1PmFVqrt21AEK06sWLrDiIK1G08otxGM4/view?usp=drivesdk @Zanthed

Zanthed, 2022-03-07:
@ChAJ07 I don't know what these logs are.

Please obtain logs from logcat through adb logcat.

https://developer.android.com/studio/command-line/logcat

thestinger, 2022-04-12:
Please try version 19.

https://github.com/GrapheneOS/Camera/releases/tag/19

CameraX implemented significant video recording portability improvements.

ChAJ07, 2022-04-12:
Th

> Please try version 19.
> 
> https://github.com/GrapheneOS/Camera/releases/tag/19
> 
> CameraX implemented significant video recording portability improvements.

Thanks It works Now❤️❤️❤️

closed 2022-04-12 by ChAJ07