GrapheneOS/Camera #137: F-Droid release

12people, 2022-03-06:
Now that the app has been release on the Google Play Store, it'd be great to also see it available on the F-Droid FOSS app store. Are there any plans to do so?

closed 2022-03-06 by randomhydrosol

12people, 2022-03-06:
@randomhydrosol Are you closing this because it's not planned or because an F-Droid release exists or is being readied? If it's the former, could you give some reasoning as to why it's not planned?

Zanthed, 2022-03-06:
There are no plans to put this on F-Droid and anyone packaging our apps on F-Droid will be asked to remove it.

12people, 2022-03-06:
OK. Is this because of the way app signing works on F-Droid, with all apps being signed with the same key? (The same reason Signal gave for not being on F-Droid.) Or is there a different reason?

thestinger, 2022-03-06:
F-Droid has numerous security issues which they have tried to cover up rather than acknowledging and improving. It also has a lot of usability and robustness deficiencies. It still targets API 25 (Android 7.1) and they still use legacy signing from before Android 7 for their metadata which has serious deficiencies along with their complete lack of support for key rotation.

F-Droid developers have chosen to act maliciously towards the GrapheneOS project. They've spread misinformation across multiple platforms and have directly engaged in harassment and bullying targeting our project members. It's a highly untrustworthy project which will be increasingly unwelcome on GrapheneOS.

We intend to add a database of signing key pins mapping app ids to their official signing keys. This will show a verification failure for apps built by F-Droid reusing developer app ids with their own signing keys, which is the vast majority of the apps in their repository. It will be possible for users to bypass, but it will become an increasingly scary warning about the app having invalid signing keys. You can see that the GrapheneOS releases through the Play Store use a different app id with a `.play` suffix in order to avoid conflicts with the releases we sign and ship ourselves through our app repository.

Use our app repository client for our apps rather than F-Droid:

https://github.com/GrapheneOS/Apps/releases

No one should be redistributing our apps via F-Droid. F-Droid will be replaced and obsoleted on GrapheneOS, and also elsewhere. Their focus is on getting themselves money through grants and contract work. They care little for the quality of their app or the security / well-being of their users. It's not going to last.