GrapheneOS/Camera #144: Crashes when no rear-camera available

astonishingriverboat, 2022-03-09:
### Camera release
16

### Brief summary
My Pixel 3a currently has no rear-camera.
And when opening the app, it crashes.
If i'm quick and switch to the existing front-camera in time, the app works as expected.

### Steps to reproduce
- Uninstall rear-camera
- Open Camera

### Expected behaviour
Automatically use existing front-camera and hide camera switching button.
And if no camera available, then display something along the lines of "No Camera Found".

### Actual behaviour
App is open for about 2 seconds, then the App crashes.
If i switch to the existing front-camera in these 2 seconds, the app works as expected.

MHShetty, 2022-04-08:
Hey @astonishingriverboat,

We've made some minor changes in the code to try resolve the issue you're currently facing. Could you please test the latest changes pushed to the repository for the same?

Thanks a lot for reporting this issue!

Regards,
Mohit

thestinger, 2022-04-09:
I can make a new build for testing.

thestinger, 2022-04-12:
Should be resolved by version 19. Please let us know if it's still an issue.

https://github.com/GrapheneOS/Camera/releases/tag/19

astonishingriverboat, 2022-04-14:
Thanks, camera and video now work as expected.
But when clicking on "QR Scan" the app still crashes.

thestinger, 2022-04-14:
QR scanning currently only uses the rear camera, since we had decided it didn't make sense to use the front camera. It could automatically fall back to using the front camera if the rear camera is unavailable. I don't know if it makes sense to support explicitly switching the camera.

astonishingriverboat, 2022-04-16:
Yeah, i also think explicitly switching the camera makes no sense in QR mode. But i would appreciate an automatic fall back to the front camera.

thestinger, 2022-04-22:
It won't work yet but with version 23 it should no longer crash.

astonishingriverboat, 2022-04-22:
I had actually noticed another early initialization bug in 21. But before i was able to report it, you already fixed it in 22.
Now in 23, i tried all the weird combinations i could think of and i can't get it to crash :).

Thank you very much for your work.

thestinger, 2022-05-11:
Going to let this be replaced by https://github.com/GrapheneOS/Camera/issues/212 since the crash is fixed.