GrapheneOS/Camera #147: Closing the app while a photo is saving causes it to be lost

austinried, 2022-03-13:
Took me a bit to figure out what was happening here, but every now and then when I open my camera to snap a quick picture, I'll end up not having the photo later.  Seems to be that if the app is closed before the photo is finished saving (which takes a second or two on my Pixel 4, GrapheneOS), the image is just lost forever.  I often open my camera by double-tapping the lock button to take a photo and then close it after it's done, or take a few and end up closing it before the last photo is saved, which leads to lost pictures.

realpixelcode, 2022-03-24:
I've got the same problem with Graphene camera but also with Open Camera.

thestinger, 2022-04-26:
Please try again with version 25.

thestinger, 2022-04-26:
You can enable the Camera Beta channel in the app repository client by going to the app page and the drop-down menu settings to get the release, or install it from GitHub.

austinried, 2022-04-26:
I've tested 25 a bit and it seems like there is now a 1-2 second lag between pressing the button and the "shutter" noise/picture actually being taken. I guess this is preferable to before when the shutter would go off right when the button was pressed but the picture wasn't actually taken until 1-2 seconds later, but it feels worse to use.

There also seems to be a new problem now where if I close the app in that 1-2 second lag period, sometimes when I open it again the camera is stuck in the "please wait until the image is saved" mode forever with the spinner around the gallery image spinning.  This lasts until I close the camera and open it again.

So it kinda seems like the core issue is still here, in that if I press the button and close the app before the image is saved, the image is lost, and it also sometimes breaks the camera the next time I open it.

Zanthed, 2022-04-26:
> I've tested 25 a bit and it seems like there is now a 1-2 second lag between pressing the button and the "shutter" noise/picture actually being taken. I guess this is preferable to before when the [...]

It's more accurate and the intended way because the shutter sound previously was misleading. Users would hear the shutter sound when they hit the button, but it wasn't when it actually took the picture. The shutter sound is now when it actually takes the picture. It now works exactly the same as Google Camera.

austinried, 2022-04-27:
Yeah I get that it's more true to when the image is saved, and therefor preferable to being misleading, but the lag still does not feel good to use.

To say it's exactly the same as Google Camera is not accurate though.  Google Camera takes the photo/sounds the shutter within probably 100ms of when you press the button, maybe even quicker.

thestinger, 2022-04-27:
Our app plays the sound and animation when the photo is captured, not when the photo is saved. It's already as fast as it's going to get via Camera2 / CameraX and there is no longer a delay for further processing and then saving the photo after it's reported to us as captured. There isn't a 1-2 second delay for capturing the photo. Are you using Quality mode or Latency mode? Please read the thread at https://twitter.com/GrapheneOS/status/1519047412052017157. You're not comparing apples to apples with other apps if you use Quality mode which waits for 3A lock.

thestinger, 2022-04-27:
There is no way to capture photos regularly than Latency mode already does. There is no room for improvement beyond CameraX working on supporting ZSL as another photo. Many widely used apps use ZSL and don't actually capture the photo when you request it but rather ask the OS to continuously take photos and then simply give you the last one they have available when you press the button. It's not quite the same as taking a new photo, especially in Quality mode where it tries to make sure auto-focus, auto-exposure and auto-white-balance are updated.

austinried, 2022-04-27:
> There isn't a 1-2 second delay for capturing the photo.

I... don't know what to say other than there definitely is? I'm in the default mode, which is Quality, and there is a delay.  You can tell by hitting the button and then moving the camera and seeing what image is taken (before moving or after moving).

I did not even know there was another mode, so I have just switched it to Latency mode per your suggestion, and I see now that while there is a delay in sound/shutter animation, there is no longer a delay in capture, which is way better.  Personally I would prefer if in Latency mode if the shutter/sound matched the capture, but this works.  This seems like it should be the default mode to me.

thestinger, 2022-04-27:
The sound is already played when the OS tells us the photo is captured. All we can know is when it's reported to us as captured, not the immediate moment that capture occurs. It's not possible.

austinried, 2022-04-27:
Ok well regardless of the delayed sound/animation, if I close the app during that delay the photo still goes missing, in both Quality and Latency modes.  Are you saying this cannot be fixed?

Additionally, there is a new issue where if I press the home button while it's saving it gets stuck in "saving" state the next time I open the app (spinner around the gallery icon in the lower right) and prevents me from taking any photos until the app is closed and opened again.  Would you like me to open a separate bug for that?

peepls, 2022-04-27:
I am having the same problem. Pixel 4a. Sometimes, if I close Camera too quickly, the picture is either not taken or not copied to the galery. In case of the latter, however it is possible to access the picture through [Device Name]/DCIM/Camera. But not through Pictures/Camera.

It is annoying if no picture at all is taken. Because often I rely on being able to take a quick picture.

thestinger, 2022-04-27:
For the time being, stop forcefully closing the app activity.

thestinger, 2022-04-27:
It does show when the image has been saved.

thestinger, 2022-04-27:
If you're closing the app while it's saving, then sometimes it will have had time to add it to the media store but not to the directory. It's entirely expected with how it works right now. It doesn't save in the background. Consider what would happen if you exited an application on your desktop while it was saving. You'd probably wait for it to finish before you closed it. It's possible for us to spawn a background service for saving but I don't think it's what people should assume from an app, and you can see the same approach as ours in other camera apps.

austinried, 2022-04-27:
My desktop is not my phone, but if I start a task and then lock the PC or minimize the app (pressing the home button essentially) I do expect it to finish, yes. Both of those scenarios are causing the image to be lost here.

thestinger, 2022-04-27:
You're implying that switching away from the app stops it from saving the photo. Is that what you're saying? Closing the app in recent apps is equivalent to minimizing and then *closing* an app on the desktop. Recent apps is the task bar equivalent where the app has been minimized. Closing it there is closing the app and killing the activity, etc.

austinried, 2022-04-27:
Yes, I am saying if the app is not explicitly in the foreground, it may not save the photo.

And as I mentioned above, when pressing the home button now there is an additional problem of the image not being saved plus you can no longer take new images because it is stuck in the saving state.

peepls, 2022-04-28:
@thestinger thank you for taking the time. 
I would like to elaborate a little bit, to hopefully make my use case more clear.

How I would like to use the app:
* See something interesting without having much time to capture it. Maybe I am on the train and have to get out at the next station, while passing by something interesting
* Grab my phone, double tapping the power button to activate the camera without unlocking the phone
* Tap the shutter once I see the preview on the screen - image is immediately captured
* Image will be saved in background without me having to wait or worry about it
* Tap power button once to lock my phone again (total time elapsed: approximately 3-5 seconds)
* store my phone in my pocket

Thank you for your astonishing work!

Zanthed, 2022-04-28:
> Tap the shutter once I see the preview on the screen - image is immediately captured

You need to use latency mode for that then. Quality mode waits for the 3A locks. In this scenario, you will not have enough time for that and need to use latency mode or the upcoming ZSL mode.

austinried, 2022-04-29:
I've switched to Latency mode and it's much better for this use case, and after noticing version 29 just came out with what looked like a fix for the other issue I noticed, I did some testing and can no longer repro the getting "stuck" in the saving state issue, which is great.

@Zanthed I'm not really familiar with ZSL mode, but do you think that will also solve the image being lost issue?

thestinger, 2022-05-19:
This appears to be working now.

closed 2022-05-19 by thestinger