GrapheneOS/Camera #149: Fetching Location Fails on Opening App

sethgillett, 2022-03-14:
I've been experiencing a bug since the app came out where if I open the app with location tagging enabled and the location permission set to "while using the app", the app will consistently fail to fetch location with the message "Fetching location... This might take a while". The location never resolves and any pictures taken are untagged.

However, if I open the app with location permission set to "ask every time" and grant access after the app opens, location resolves correctly and pictures are correctly tagged.

I assume this has something to do with the way location fetching is handled in [CustomLocationListener.kt](https://github.com/GrapheneOS/Camera/blob/main/app/src/main/java/app/grapheneos/camera/CustomLocationListener.kt), and I was wondering if anyone has also experienced this bug or has any ideas since I'm pretty new to android development.

lbdroid, 2022-03-17:
Part of this issue is probably the fact that google likes to silently track your physical location by performing GPS location checks every few minutes. This keeps the GPS hardware perpetually primed so its easy and fast to grab a location when needed. GrapheneOS, on the other hand, does NOT constantly track your physical location, therefore it can take a while to get coordinates. But taking pictures is very time sensitive, so there may not be enough time to do that, in which case, it just has to save the file and move on.

thestinger, 2022-03-17:
This is about the geotagging implementation in our Camera app, which we know hasn't been fully implemented yet and needs to be finished. It has nothing to do with Google. GrapheneOS works the same way as AOSP with a perfectly good implementation of GNSS + A-GPS. It doesn't have a network location service built into the OS like the stock OS or iOS which would help a lot with getting a much quicker, lower power usage location estimate and indoor location detection. That probably isn't something we want to provide though. We plan on providing a "network" location service based on a local database. There's an issue about it on the GrapheneOS OS issue tracker. It's really not relevant to his because it's not why the Camera app doesn't reliably obtain location from the OS. We just need to finish the implementation of getting the location.

thestinger, 2022-03-17:
Removing our comments about this because it's off-topic and should be discussed in the Matrix chat rooms or elsewhere rather than here.

MHShetty, 2022-03-25:
Hey @sethgillett,

@empratyush has worked on re-implementing geo-tagging with a different approach and it seems to work well on our end. Could you please pull the latest changes made to our repository and test the app for the same?

Thanks a lot for trying out our camera app and reporting this issue!

Regards,
Mohit

thestinger, 2022-03-26:
There's apparently a crash with stock OS where Play services has network location accuracy disabled, likely because the network location provider throws / reports an error which we aren't handling and should be.

thestinger, 2022-03-26:
This is implemented by version 17.

closed 2022-03-26 by thestinger

sethgillett, 2022-03-26:
> Hey @sethgillett,
>
> @empratyush has worked on re-implementing geo-tagging with a different approach and it seems to work well on our end. Could you please pull the latest changes made to our [...]

Just pulled the latest changes, and the issue appears to be fixed!