GrapheneOS/Camera #152: No tap-to-focus in QR code mode

realpixelcode, 2022-03-25:
In the regular photo mode it's possible to adjust the focus to a specific area by tapping there. However, the QR code mode only supports automatic focus which (at least on my device) always adjusts to the background instead of the foreground where the QR code is.

So, the QR code mode should also support "tap-to-focus".

thestinger, 2022-03-25:
The focus for QR mode is done with the scanning square as a region. Tap-to-focus simply sets a point as the region. If your device doesn't have working auto-focus for a larger region then that's really something which should be worked around in CameraX by making it only set a point instead. Which device is it?

closed 2022-03-25 by thestinger

realpixelcode, 2022-03-25:
> This isn't planned. The device needs a CameraX workaround if auto-focus doesn't work correctly. You'll need to provide information on the device and OS.

I'm using the app on my Fairphone 3 with /e/ OS Android 11. Unfortunately, scanning small QR codes printed on the side of a narrow product box is quite tricky.