GrapheneOS/Camera #156: Files deleted with Google Photos (moved to trash) are being shown in in-app gallery

jorgeluiscarrillo, 2022-03-28:
Google Photos / Gallery Go have trash functionality. I am not sure where in the filesystem these files are stored when they are sent to the trash, but Camera is still showing them. Only once the files are deleted from the trash do they not show up in the in-app gallery. I have only tested with these apps so I am not sure how it works with other gallery apps with a built-in trash.

thestinger, 2022-03-28:
We could support the standard MediaStore trash functionality added in Android 11:

https://developer.android.com/reference/android/provider/MediaStore.MediaColumns#IS_TRASHED

I don't know if they're actually using the standard infrastructure for the feature. We could also optionally move the captured media to the trash when deleted in the camera app but we would need to add a way of showing the ones in the trash and permanently deleting them, and it might be bad as a default for a privacy-focused app, although it's not as if deleting individual files is actually reliable.

thestinger, 2022-03-28:
We could support the standard MediaStore trash functionality added in Android 11:

https://developer.android.com/reference/android/provider/MediaStore.MediaColumns#IS_TRASHED

I don't know if they're actually using the standard infrastructure for the feature. We could also optionally move the captured media to the trash when deleted in the camera app but we would need to add a way of showing the ones in the trash and permanently deleting them, and it might be bad as a default for a privacy-focused app, although it's not as if deleting individual files is at all guaranteed to make them unrecoverable.

locuturus, 2022-05-23:
Google Photos renames trashed files as ```.trashed-[10-digit-numeral]-[original filename]``` and keeps them in the original directory.

PR #298 (2022-10-10): don't show trashed item in `InAppGallery`
fixes #156
closed 2022-10-10 by thestinger