GrapheneOS/Camera #159: Edit captured photo before it's saved

steve47457, 2022-03-31:
How can I programmatically edit a captured photo before it's saved? For some reason, `onCaptureSuccess`  in the `takePicture()` method (line 116 - CaptureActivity.kt) is never called. Can someone provide clarification? Thanks.

empratyush, 2022-04-01:
IMO you can do it on [ImageCapturer.kt](https://github.com/GrapheneOS/Camera/blob/main/app/src/main/java/app/grapheneos/camera/capturer/ImageCapturer.kt#L96)

steve47457, 2022-04-01:
@empratyush Thank you for your reply. Can you please give an example? Because ImageCapturer.kt only has a `OnImageSavedCallback`.

empratyush, 2022-04-01:
you need edit saved image afaict.

steve47457, 2022-04-01:
@empratyush There's no way to edit the photo **before** it's saved?

steve47457, 2022-04-01:
@thestinger What do you think?

thestinger, 2022-04-01:
You should ask in the CameraX discussion group.

closed 2022-04-01 by thestinger

steve47457, 2022-04-01:
@thestinger I think the issue is with your library because the CameraX functionality already exists (`OnImageCapturedCallback()`)

thestinger, 2022-04-01:
We don't make a library.

steve47457, 2022-04-02:
I meant the GrapheneOS Camera app. Is there way to treat the captured photo before it's saved? @thestinger

steve47457, 2022-04-02:
In the same way the GrapheneOS Camera app can increase/decrease the brightness of a photo, is it not possible to edit the photo in other ways? @thestinger

thestinger, 2022-04-02:
It doesn't have a way to increase/decrease brightness. It has a way to control the exposure settings with an exposure compensation slider. That's changing how much light is absorbed by the image sensor. It's much different than adjusting brightness and it isn't done after the photo is taken but rather as part of taken it.