GrapheneOS/Camera #169: Fdroid

Donkey-Doug, 2022-04-03:
Is there an fdroid repo? Or are there plans to add grapheneos apps to fdroid?

akc3n, 2022-04-03:
No, there is not. And no, there is no plans, nor will there be any plans to add anything to F-Droid.

closed 2022-04-03 by Donkey-Doug

thestinger, 2022-04-03:
Our Camera app is already in our own app repository. We're working on developing a far more robust, secure and usable alternative to F-Droid built around the modern platform including unattended updates, split APK support and the Android 6+ permission model based on runtime toggles which has gotten substantially better over time.

https://github.com/GrapheneOS/Camera/releases

thestinger, 2022-04-03:
F-Droid isn't suitable for usage by GrapheneOS. It has serious problems and also highly untrustworthy people with commit access and access to their builds / signing keys who have already demonstrated that they're willing to do highly underhanded and malicious things to benefit themselves. F-Droid has to be replaced.