GrapheneOS/Camera #177: UI - Hole punch camera on top of settings button (Moto G Power 2020)

JoeMamaAME, 2022-04-11:
Device: Moto G Power 2020

Issue: The hole punch camera covers most of the settings dropdown icon, only some of the gray on the sides can be seen. It's still tapable of course.

thestinger, 2022-04-11:
This is because your device implements the vertical space partially covered by the notch available to apps in a non-Android-compatible way not properly implementing the API for detecting the usable space. Our app does use the API for detecting the usable space correctly and this isn't an issue on compliant devices. There's not much we can do about Motorola having a broken configuration option.

thestinger, 2022-04-11:
It's a bug in the OS for the device. Xiaomi had a similar bug and fixed the issue as part of their port to Android 12. The most that can be done is reporting it to Motorola because we won't be hard-wiring an internal database with their device models and the notch location. They need to implement Android APIs correctly. This is likely an issue due to enabling a non-default configuration option which isn't Android compliant even though it should be and would still work fine if it was and the API returned the proper values.

closed 2022-04-11 by thestinger