GrapheneOS/Camera #178: "Optimize for Quality" doesn't work on AOSP

CaseyBakey, 2022-04-12:
Hello there,

Using latest GrapheneOS Camera (r18) on Google Pixel 3 running AOSP (SP1A.210812.016.C1).

When I'm trying to take photos with the default "quality" setting, nothing works.

The preview keeps loading, and I can't click on it ("Please wait for the image to get captured before trying to open the gallery").

Starting another app, and going back to Camera, I then can click preview and am welcomed with a full black image.

Here is the error I got from ImageView in ADB when trying to do that:
'''
Unable to open content: content://com.android.externalstorage.documents/tree/primary%3ADCIM/document/primary%3ADCIM%2FIMG_20220412_082510.jpg
android.graphics.ImageDecoder$DecodeException: Failed to create image decoder with message 'unimplemented'Input contained an error.
	at android.graphics.ImageDecoder.nCreate(Native Method)
	at android.graphics.ImageDecoder.createFromAssetFileDescriptor(ImageDecoder.java:351)
	at android.graphics.ImageDecoder.access$400(ImageDecoder.java:172)
	at android.graphics.ImageDecoder$ContentResolverSource.createImageDecoder(ImageDecoder.java:286)
	at android.graphics.ImageDecoder.decodeDrawableImpl(ImageDecoder.java:1766)
	at android.graphics.ImageDecoder.decodeDrawable(ImageDecoder.java:1759)
	at android.widget.ImageView.getDrawableFromUri(ImageView.java:1017)
	at android.widget.ImageView.resolveUri(ImageView.java:986)
	at android.widget.ImageView.setImageURI(ImageView.java:561)
	at androidx.appcompat.widget.d0.setImageURI(Unknown Source:0)
	at j2.q.B(Unknown Source:59)
	at j2.q.e(Unknown Source:4)
	at app.grapheneos.camera.ui.activities.MainActivity.onResume(Unknown Source:13)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1480)
	at android.app.Activity.performResume(Activity.java:8191)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4751)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4794)
	at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:54)
	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2214)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7842)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
'''

But if I chose "latency" setting instead, it works as intended. Instant capture, and photo is viewable in the built-in gallery.

Do you have any idea of what could go wrong?

thestinger, 2022-04-12:
It sounds like the AOSP build wasn't done properly and has broken HDR+ support. Using their drivers doesn't work properly. You should use either android-prepare-vendor or adevtool.

closed 2022-04-12 by thestinger

CaseyBakey, 2022-04-17:
android-prepare-vendor was used to properly extract binary blobs from the OTA images.

As a matter of fact, Google Camera is working fine on this Pixel 3 (blueline), even with HDR mode activated.

I also did an AOSP build for sunfish (Pixel 4a) the same way, and GrapheneOS camera works just fine on this one.

So I'm no sure the problem lies here?

Maybe it's more linked to Pixel 3 (blueline)?

thestinger, 2022-04-17:
It works fine on GrapheneOS and AOSP for the Pixel 3.