GrapheneOS/Camera #180: support capturing new images while one is being saved

thestinger, 2022-04-12:


thestinger, 2022-04-26:
Implemented.

closed 2022-04-26 by thestinger

austinried, 2022-04-27:
I had imagined the solution to this would be related to #147, saving the image in a background task or something so it can't be interrupted, is this in version 25 on the beta channel as well?  I'm trying to take two images in a row and I still get the "please wait" message.

thestinger, 2022-04-27:
There is no wait after the photo is captured on version 25. You still can't take a photo before the previous one you asked to capture is captured and will never be able to do that. If you want to speed up capture itself, you need to use Latency mode, and there is nothing for us to optimize there beyond CameraX adding support for ZSL, which likely wouldn't be used by Quality mode.